SpacehuhnTech/esp8266_deauther #1148: Only one SSID on 60 same named SSID beacon

glblduh, 2019-09-15:
When I do a beacon attack with multiple SSID with the same name, only one shows up. I don't know how this occurs I was just running a beacon attack but suddenly it only shows one.

There's no error in the log.

# attack beacon
Start attacking
[Pkt/s] All:  111 | Deauths:   0/0   | Beacons:   0/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0  
[Pkt/s] All:  600 | Deauths:   0/0   | Beacons: 600/600 | Probes:   0/0

chunun, 2020-03-03:
Me too. What's the matter, please

spacehuhn, 2020-03-03:
All beacon frame with the same name appear for a scanner (like your phone or computer) as the same network. Like with big mesh networks (for example in schools, malls or a company/office space) you will see their network in your list once instead of every single access point, although there are many broadcasting the same network around you.

To make the same network name appear twice, you can add spaces to the SSID. It will get recognized as a different network (since the name is actually different) but for a human it looks just the same.

closed 2020-06-23 by stale[bot]