SpacehuhnTech/esp8266_deauther #1149: Is there an Android App that runs on mobile phones . Please give the correct name for it . Thanks

tytower, 2019-09-20:
```Is there an Android App that runs on mobile phones . Please give the correct name for it . Thanks```

<details>
  <summary> Compile log (...)</summary>

  

</details>
<br><br><br>

jLynx, 2019-09-20:
Yep, it's called Chrome. You can download it from the store but it normally comes on phones these days, so you may already have it installed

closed 2019-09-20 by jLynx

tobozo, 2019-09-20:
I hate to be that guy but we do not live (yet) in a Chrome-only world, and it's still possible to keep concerns separated with another Android-app-that-runs-on-mobile-phones called Firefox (keymaster / gatekeeper, there is only xul)

tobozo, 2019-09-20:
This project used to be maintained but I'm not sure what the requirements are :

https://github.com/ExploiTR/DeAutherDroid

jLynx, 2019-09-20:
So you are saying you do t have a browser on your phone?.....

Also any Linux distro has a browser by default

tytower, 2019-09-21:
jLynx--Mate if I choose not to run a browser on my phone that's my business . You can read . That's what I said . Why do you question someone doing things differently to you?
My phone is rooted and I am root.

Why have you closed this issue? I'm still looking for a sensible answer
Thanks tobozo I'll check that out.

jLynx, 2019-09-21:
I still can't tell if you are a troll or actually serious 😳😂

tytower, 2019-09-21:
Well mate give me the benefit of the doubt . You should not be doing this if you cannot answer a civil question. 
The questions are raised by using the device for the first time and I am not able to find any similar questions so what else can I do but ask . Look at the issue you have closed . An answer came that I was able to follow up for an android phone . So how about you piss off for a while and give someone else knowledgeable a chance to address the questions I ask

davedarko, 2019-09-21:
The reason for closing it seems obvious. There was one person guiding you towards a project and another for saying there is no dedicated app for the website the deauther is using. Then another person told you that there are other browsers, you don't have to use chrome. I'd say anything further is too off-topic from the deauther project.

tytower, 2019-09-21:
Quote "  I'd say anything further is too off-topic from the deauther project." 
What crap you utter !
Pig Headed and bigoted it seems . Why don't you give the questions a chance.? 
One answer was forthcomming . Perhaps there are others . 
I have seen an instructable that mentioned one but can't find it now.
Come on Mate stop the adamant approach and maybe take a holiday or something . You clearly have lost patience doing it here.

tobozo, 2019-09-21:
don't you dare name the utters on caturday !