SpacehuhnTech/esp8266_deauther #1150: Change name and password -made an error

tytower, 2019-09-21:
```I changed the password to a 6 letter one in error . The chip accepted that but trying to connect it needs an 8 digit one . So I tried recompiling and uploading the original code again but it does not seem to change the SSID or the password. How can I completely reset and get back to standard?```

<details>
  <summary> Compile log (...)</summary>

  

</details>
<br><br><br>

tytower, 2019-09-21:
I looked further and recompiled it another time . I found the SSID had not been changed but the password had been changed back to deauther and I was able to reconnect and correct my error .
Strange that the SSID was not changed back on reloading the code though?
Thanks.

underwoodblog, 2019-09-21:
Check the ESP8266 docu how the storage of Wifi credentials works.

1bjames1, 2019-10-03:
I also have found this to be a problem , I changed the password from default and now cannot get connected. Recompiled and reuploaded and nothing changed . Put a wifi acess point on the same nodeMCU and was able to connect to it fine but put deauther back on it and it dropped right back where it was with name and unable to connect to it . What is happening?

underwoodblog
Check the ESP8266 docu how the storage of Wifi credentials works.
Where do i do that and can you explain a bit more please . Thats a pretty vague answer.

1bjames1, 2019-10-03:
> Check the ESP8266 docu how the storage of Wifi credentials works.
Where do i do that and can you explain a bit more please . Thats a pretty vague answer.

underwoodblog, 2019-10-04:
I was to lazy to write a complete summary how the esp connection and wifi storage works, because English is not my native language and this is not an EPS8266 support forum. So I gave you a hint where to start your search. Same as you where to lazy to google it yourself:

https://lmgtfy.com/?q=how+stores+the+esp8266+wifi+password

The second search result 

https://arduino.stackexchange.com/questions/52059/does-the-esp8266-somehow-remember-wifi-access-data

If all people in this repro are as kind as you, I'm not writing any suggestions or help here again.

1bjames1, 2019-10-04:
Well the first link was just being a smart arse , wasn't it.
The second give slightly more insite but of course does not answer MY question.
My suggestion to you is is you don't want to answer questions ,dont.
I worked on understanding and interpreting  the ESP8266 chip when it first came out and I still don't understand a lot of what it does. 
Not many people do .
Where else am I likely to get answers . Not from bloody google thats for sure.

1bjames1, 2019-10-04:
Now in case someone on here has the knowledge it looks like the SSID and password might be written into the esp8266's EEROM area the first time and reloading the deauther program does not affect that memory area so a way must be found to get to the actual memory data and erase it . If I find out how I will post it back here

tobozo, 2019-10-04:
if you count on others to build your experience, it's no surprise you get frustrated by their answers

most people here are doing support on their free time so please be thankful, this issue tracker is not a sales bureau

closed 2019-10-04 by tobozo