SpacehuhnTech/esp8266_deauther #1157: ESP32  module on NodeMCU32  ESP-32S v1.1

1bjames1, 2019-10-04:
I have a nodeMCU32 I have added these boards at

http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://dl.espressif.com/dl/package_esp32_index.json
http://phpsecu.re/esp8266/package_deauther_index.json
https://dl.espressif.com/dl/package_esp32_index.json
https://dl.espressif.com/dl/package_esp32_dev_index.json
https://dl.espressif.com/dl/package_esp32_index.json
 
On compile I get 

 ```Build options changed, rebuilding all
esp8266_deauther:12:30: error: user_interface.h: No such file or directory
   #include "user_interface.h"

                              ^
compilation terminated.
exit status 1
user_interface.h: No such file or directory
```

Can anyone point me towards a solution ?

tobozo, 2019-10-04:
hey, thanks for your feedback.

you can delete this source : http://phpsecu.re/esp8266/package_deauther_index.json and use this source instead: https://raw.githubusercontent.com/wiki/tobozo/Arduino/package_deauther_index.json

Also when selecting the esp8266 board, make sure you pick it from the esp8266_deauther section as described in the [wiki](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide).

closed 2019-10-04 by tobozo

1bjames1, 2019-10-05:
Thank you for that .
I did as you suggested but that did nothing . I see you have closed this issue thinking that would fix it but it did not? No change whatever ?I picked the right board as far as I am aware.

tobozo, 2019-10-06:
I closed the issue after doing a clean install and verifying the problem came neither from the Arduino installation, nor the deauther SDK.

Since the problem isn't related to this project, the issue doesn't need to be kept open.

The symptoms you experience are caused by selecting an esp8266 board from the wrong part of the menu.

The devil is in the details, so you just need to check every one of them twice instead of questionning the documentation.

happy hacking :-)

1bjames1, 2019-10-06:
You are not getting it right . 
Issues on github are for the discussion of problems by likeminded individuals over the same software to try to work out what is happening..
You don't just close an issue because you think it does not concern you ?
"I have looked at it and determined it must be something you did" ??

What a stupid approach to solving a problem. ...NOT solving it.
Who's got control of you idiots.

I have selected the only board in the deauther section that meets the description.
The ESP8266 device has ESP WROOM 32 marked on the front of it . 
Which board should I be selecting at least ?

underwoodblog, 2019-10-06:
Can somebody block him!

tobozo, 2019-10-06:
@1bjames1 thanks for sharing your point of view on this repository, I hope you feel better now that your anger is out.

spacehuhn, 2021-12-12:
Hi! There are many issues on this repo about ESP32 support, so I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want. I appreciate your understanding.