SpacehuhnTech/esp8266_deauther #1159: can i connect lcd 240x180 to deauther board ?

sro2, 2019-10-06:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)

<details>
  <summary> Compile log (...)</summary>

  
can i connet lcd 240x180 on my deauther esp8266 .plz help if it can be possible . if it can possible then please tell how can i connect . 
</details>
<br><br><br>

sro2, 2019-10-06:
help anyone, please

tobozo, 2019-10-06:
@sro2 if you're asking about the feasability, many technical informations are missing in your question. 

You can eventually post in a more general context on an ESP8266 forum ([reddit](https://www.reddit.com/r/esp8266/) or [gitter](https://gitter.im/esp8266/Arduino)) where the community is more compatible with entry-level questions.

Closing this issue as it's not a deauther issue, but feel free to reopen and/or add some followup + technical details about your research.

closed 2019-10-06 by tobozo