SpacehuhnTech/esp8266_deauther #1160: CHANCE TO REVIEW YOUR FOOLISHNESS.

leonierocks, 2019-10-07:
CHANCE TO REVIEW YOUR FOOLISHNESS.

You are not getting it right .
Issues on github are for the discussion of problems by likeminded individuals over the same software to try to work out what is happening..
You don't just close an issue because you think it does not concern you ?
"I have looked at it and determined it must be something you did" ??

You don't just stick a label on it and forget it.

What a stupid approach to solving a problem. ...NOT solving it.
**Who's got control of you clowns. Who is overseeing the project? NO-ONE IT SEEMS !**

I have selected the only board in the deauther section that meets the description.
The ESP8266 device has ESP WROOM 32 marked on the front of it .
Which board should I be selecting at least ?

spacehuhn, 2019-10-07:
wow

underwoodblog, 2019-10-07:
Lol, 1bjames1 creating only a new account to post her again. What a madlad!

tobozo, 2019-10-07:
confusing the effects of a locked post with a banned account totally corroborates the fact that frustration can obfuscate neural paths.

digmorepaka, 2019-10-07:
I have looked at this issue, and determined it's a waste of time for both the developers, maintainers, and you. If you don't like how the issues section meant for *Bug reports* is handled,  no one forces you to make a post or read the others.

lbussy, 2019-10-07:
Clearly, the gentleman deserves a refund?   Twice what he paid for all of his troubles!

tobozo, 2019-10-07:
@digmorepaka I'll do you one better: "...it's a waste of time for *throth* the developers, maintainers, and you"

PwnKitteh, 2019-10-07:
I think I deserve a refund for my time for reading this stupid ass post

f0wl, 2019-10-09:
That's like buying an iPhone and complaining that it doesn't run Android

jasonjpeters, 2019-10-24:
> That's like buying an iPhone and complaining that it doesn't run Android

I'll do ya'll one even more better -- why is Android.

astonge, 2019-10-24:
You win. I quit.

jLynx, 2020-02-24:
Closed as OP seems to be off his meds.

closed 2020-02-24 by jLynx