SpacehuhnTech/esp8266_deauther #1161: Adding Functions

Helixable, 2019-10-09:
Hello! I recently discovered the world of Wifi Deautherization. I ordered a WeMos OLED ESP8266 Module. I had a bit of trouble flashing the .bin file, but I managed to do it using the ND tool. I love the functions, but I would like to see more. Instead of asking the devs to add more, I was hoping somebody could explain how to add more functions. Like "Ping" or "DOS" or even stress test instead of buffering the actual network. A favor someone could do us is make a framework for commands. Thanks!

tobozo, 2019-10-10:
@Helixable  thanks for your feedback, 

This probably won't happen on esp8266_deauther as the current code base already uses most of the ESP8266 resources.

I'm labelling this post as feature request and closing but feel free to reopen if you manage to implement it without pulling your hair off :-)

closed 2019-10-10 by tobozo

Helixable, 2019-10-10:
So basically you're saying the ESP can't handle any more functions?

tobozo, 2019-10-11:
I'm saying it's unlikely the *deauther* can handle any more functions, resource wise, this means hardware (alghough not limited to SPIFFS but heap size) as much as human (would you code it?).

Helixable, 2019-10-11:
If I knew C / Arduino I would try to code it. Sorry if it was a stupid
question. I code in different languages. Thanks for all you devs for this
amazing tool. 🙂

tobozo, 2019-10-11:
Asking about including more stress tools to the existing tool isn't stupid at all, just a bit out of scope (deauther is for security assessment on 2.4GHz WiFi networks).

Also you don't need to know C to upload some extra html files into the SPIFFS partition, provided you know HTML/CSS/JS enough to maintain those.

Helixable, 2019-10-11:
Thanks. I know a decent amount of Html, Css, and Js but I have no idea how
to implement them into the deauther. I don't even know what a SPIFFS
partition is.

tobozo, 2019-10-11:
SPIFFS is where the deauther HTML/CSS/JS files are stored, it's just a part of the flash memory that's used a a filesystem.

When the deauther runs for the first time, the partition is formatted and necessary files are created, but they can be erased/overwrittent afterwards.

this may help: https://randomnerdtutorials.com/install-esp8266-filesystem-uploader-arduino-ide/

problems you will encounter : chicken & egg situation where uploading your own files deletes the existing files, better practice on a basic WebServer example so you can understand first how it works, and get help from a more generalistic forum such as [gitter](https://gitter.im/esp8266/Arduino)

happy hacking !

Helixable, 2019-10-11:
Oh ok. Thanks!

Helixable, 2019-10-13:
Quick question. So when I flashed my deauther, I used a flashing tool that
you could choose a custom .bin file. How would I make a .bin file after I
made the modified version of the HTML files?

tobozo, 2019-10-14:
you only need to overwrite the SPIFFS part and this isn't done using a binary but using the [Filesystem Uploader in Arduino IDE](https://randomnerdtutorials.com/install-esp8266-filesystem-uploader-arduino-ide/)

Helixable, 2019-10-14:
Oh ok. This means I can just plugin the flashed deauther and upload the new
spiffs using the instructions on the Link you gave me?

tobozo, 2019-10-14:
as suggested before, more answers to generalistic questions on the [gitter](https://gitter.im/esp8266/Arduino)

also try it before you ask it :-)

Helixable, 2019-10-14:
Ok then. 🙂