SpacehuhnTech/esp8266_deauther #1165: Using blank space in station SSID and password FIX

BorcheM, 2019-10-18:
It was bugging me that in the web interface in the settings tab, if you using two words with space between, it was truncated after the space.
For example if you set the SSID to "My Deauther" and reboot, than the SSID is really set to "My Deauther", but in the web interface it only show "My". In the console it was showing "My Deauther", like it should be. So setting  was fine, but loading was problem.

To fix this I edited the settings.js file on line 26:

> value="+settingsJson[key]+"

I replaced it with:
> value='"+settingsJson[key].toString()+"'

After that I was using the web converter to create new web page.
I loaded blank sketch, and when I try to upload "esp8266_deauther.ino", new error when compiling.

> 'class Settings' has no member named 'getWebSpiffs'

Solution to this was to edit webfiles.h on line 36:

> if(settings.getWebSpiffs())

to:
>if(settings.getWebSettings().use_spiffs)

Or you can edit the script webConverter.py on line 225:

> f.write("if(settings.getWebSpiffs()){\n")

to:
>f.write("if(settings.getWebSettings().use_spiffs){\n")

I tested it and all seams to be working now.

spacehuhn, 2019-10-19:
Would you like to make a pull request for the changes, so I can just merge it?

https://help.github.com/en/articles/creating-a-pull-request

spacehuhn, 2020-06-16:
Merged in https://github.com/SpacehuhnTech/esp8266_deauther/pull/1166

closed 2020-06-16 by spacehuhn