SpacehuhnTech/esp8266_deauther #1171: Feature request: Make serial console play nice(r) with terminal programs

sgofferj, 2019-10-26:
Version 2.1.0 from https://github.com/spacehuhn/esp8266_deauther/releases/tag/v2.1.0

The serial console seems to have some kind of timeout after which it will take the input buffer and evaluate it. That doesn't play nice with a normal terminal program because one has to type quite fast and can't have any short pauses when typing, otherwise, the console will grab what has been written to that point and take it as a command.
It would play nicer with terminal programs if that timeout could be disabled in favor of <CR> or <CR><LF> input.

Ozymandias1700, 2020-05-16:
I know this might be a long shot given that this is an old post but how exatly did you even get it to work with a regular serial connection? I managed to flash the firmware no bother after figuring out the wiring for flash mode, but now when I do
`screen /dev/ttyAMA0 115200`
everything looks okay, I get all the normal info and the scans etc but it won't let me type anything into the screen serial connection. Am I doing something wrong?

sgofferj, 2020-05-23:
@Ozymandias1700 
the console doesn't provide any echo.

Ozymandias1700, 2020-05-25:
I know, it simply wasn't working the first few times. Randomly started working tho

spacehuhn, 2020-05-27:
Hey, sorry for not reacting to this earlier. The same issue has been on my radar too. 
The new version I am working on (https://github.com/SpacehuhnTech/esp8266_deauther/tree/v3) shouldn't timeout anymore and I'd be happy if you could check it.

Also we just started working on this project: https://github.com/SpacehuhnTech/Huhnitor
A dead simple serial terminal made specifically for the deauther. That should do the trick for the old/current deauther version.

Ozymandias1700, 2020-05-30:
This looks fantastic, thanks. My own little script I wrote for the issue was working but definitely lacking! I'll give it a look asap, thanks for all the great work!