SpacehuhnTech/esp8266_deauther #1172: Feature request: Capacitive touch buttons

LowTrunkOzz, 2019-11-02:
I've searched and not exactly found an answer. I'm trying to make Capacitive touch buttons work. Should I just use the simpleButton library and define updated buttons in there? I haven't gotten my capacitive buttons in yet, so I haven't been able to test. Will advise after receiving them. Thanks in advance!

tobozo, 2019-11-03:
you may want to subscribe to [this post](https://github.com/spacehuhn/esp8266_deauther/issues/1084), and add your followup

LowTrunkOzz, 2019-11-03:
I saw that post but I'm adding 3 capacitive buttons alone, not through an i2c module, so I wasn't exactly sure if functions would cross. These are the buttons I ordered: https://www.amazon.com/dp/B01LWKFS7L/ref=cm_sw_r_sms_apa_i_2YKVDbG210S9F

tobozo, 2019-11-03:
oh right, at first I thought you were talking about buttons on a TFT with capacitive touch.

This brings back the challenge into the [SimpleButton](https://github.com/spacehuhn/SimpleButton/blob/master/src/SimpleButton.h) library perimeter anyway.

LowTrunkOzz, 2019-11-06:
Got my capacitive sensors today and quickly wired one to see what happens. I didn't change anything in the code, just hooked one up to the pin for the down button. It does register, but without touching the pad it scrolls continuously until I touch the pad, then it stops until i move my finger away from the sensor at which time it continues on scrolling. I'm assuming the code will need changed from INPUT_ PULLUP to just INPUT with pinMode but that's just a guess at this point. I'll get to tinkering when I have a day off.

LowTrunkOzz, 2019-11-10:
Buttons now work. I ended up editing the following section of DisplayUI.cpp:

void DisplayUI::setupButtons() {
    up   = new ButtonPullup(BUTTON_UP);
    down = new ButtonPullup(BUTTON_DOWN);
    a    = new ButtonPullup(BUTTON_A);
    b    = new ButtonPullup(BUTTON_B);

to look like this:

void DisplayUI::setupButtons() {
    up   = new Button(BUTTON_UP, false);
    down = new Button(BUTTON_DOWN, false);
    a    = new Button(BUTTON_A, false);
    b    = new Button(BUTTON_B, false);

and now up, down, and a all work.

tobozo, 2019-11-10:
thanks for sharing your results :+1: 

you are now at risk of butt-deauthing :rofl:

closed 2019-11-10 by LowTrunkOzz