SpacehuhnTech/esp8266_deauther #1175: pwned authentication error

danyboy1972, 2019-11-07:
i flashed my wemos, plugged it into a usb cable and get the pwned SSID but when i type the password i get 'authentication error' does anyone have any ideas whats wrong?
cheers.

Jacob12138xieyuan, 2019-11-29:
what is the password?

danyboy1972, 2019-12-09:
hello,
the password is deauther.

spectrapulse, 2019-12-11:
I've tried using the password "deauther" but it refuses to authenticate. Even after reflashing it refuses to do so.

forseenink3238, 2020-01-09:
Are you doing this on a laptop.

spectrapulse, 2020-01-09:
Yes...

forseenink3238, 2020-01-09:
Try to connect to pwned using a phone, this time there shouldn't be any authentication problem.

spectrapulse, 2020-01-09:
I already tried that but the same thing happens on my phone. I've tried using multiple Android powered phones and an iPhone of a friend.

mmuyakwa, 2020-02-24:
If you changed the password (in code or on the websites settings) and it is shorter than 8 characters, you won't be able to log in anymore.
It is highlighted nowhere which is unfortunate.

closed 2020-09-21 by stale[bot]