SpacehuhnTech/esp8266_deauther #1177: The diode on the board flashes every 5 seconds.

88cretin88, 2019-11-13:
Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)

tobozo, 2019-11-13:
Hello, what do you get in the Serial output? Have you tried erasing the flash and flashing it again?

[image]

88cretin88, 2019-11-13:
> Здравствуйте, что вы получаете в последовательном выходе? Вы пробовали стирать вспышку и мигать ее снова?
> 
> [image]

After flashing everything works fine, it stops working the next day. I don’t know why this happens.

altimmons, 2019-12-07:
That’s roughly the time it would take for the micros register to overflow and reset. If someone were using that it could be a cause.

closed 2020-08-18 by 88cretin88