SpacehuhnTech/esp8266_deauther #1180: no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'

the-mcnaveen, 2019-11-18:
<details>
  <summary> no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'</summary>

Arduino: 1.8.10 (Linux), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 80MHz, DIO, 115200, 1M (256K SPIFFS)"

sketch/Names.cpp: In member function 'void Names::load()':
Names.cpp:10:38: error: no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'
     DynamicJsonBuffer jsonBuffer(4000);
                                      ^
sketch/Names.cpp:10:38: note: candidates are:
In file included from /home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:7:0,
                 from /home/ninja/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:13,
                 from sketch/Names.h:10,
                 from sketch/Names.cpp:1:
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note: ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer()
   DynamicJsonBuffer();
   ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note:   candidate expects 0 arguments, 1 provided
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:23:7: note: constexpr ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(const ArduinoJson::DynamicJsonBuffer&)
 class DynamicJsonBuffer : public JsonBuffer {
       ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:23:7: note:   no known conversion for argument 1 from 'int' to 'const ArduinoJson::DynamicJsonBuffer&'
sketch/Settings.cpp: In member function 'void Settings::load()':
Settings.cpp:9:38: error: no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'
     DynamicJsonBuffer jsonBuffer(4000);
                                      ^
sketch/Settings.cpp:9:38: note: candidates are:
In file included from /home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:7:0,
                 from /home/ninja/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:13,
                 from sketch/Settings.h:9,
                 from sketch/Settings.cpp:1:
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note: ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer()
   DynamicJsonBuffer();
   ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note:   candidate expects 0 arguments, 1 provided
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:23:7: note: constexpr ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(const ArduinoJson::DynamicJsonBuffer&)
 class DynamicJsonBuffer : public JsonBuffer {
       ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:23:7: note:   no known conversion for argument 1 from 'int' to 'const ArduinoJson::DynamicJsonBuffer&'
sketch/Settings.cpp: In member function 'String Settings::getJsonStr()':
Settings.cpp:124:38: error: no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'
     DynamicJsonBuffer jsonBuffer(4000);
                                      ^
sketch/Settings.cpp:124:38: note: candidates are:
In file included from /home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:7:0,
                 from /home/ninja/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:13,
                 from sketch/Settings.h:9,
                 from sketch/Settings.cpp:1:
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note: ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer()
   DynamicJsonBuffer();
   ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note:   candidate expects 0 arguments, 1 provided
sketch/SSIDs.cpp: In member function 'void SSIDs::load()':
SSIDs.cpp:9:38: error: no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'
     DynamicJsonBuffer jsonBuffer(4000);

                                      ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:23:7: note: constexpr ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(const ArduinoJson::DynamicJsonBuffer&)
 class DynamicJsonBuffer : public JsonBuffer {
       ^
sketch/SSIDs.cpp:9:38: note: candidates are:
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:23:7: note:   no known conversion for argument 1 from 'int' to 'const ArduinoJson::DynamicJsonBuffer&'
In file included from /home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:7:0,
                 from /home/ninja/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:13,
                 from sketch/SSIDs.h:10,
                 from sketch/SSIDs.cpp:1:
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note: ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer()
   DynamicJsonBuffer();
   ^
/home/ninja/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:25:3: note:   candidate expects 0 arguments, 1 provided
[... 91 lines cut ...]
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/EEPROM
Multiple libraries were found for "ArduinoJson.h"
 Used: /home/ninja/Arduino/libraries/ArduinoJson
 Not used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/ArduinoJson
Multiple libraries were found for "SH1106Wire.h"
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0
Multiple libraries were found for "SimpleButton.h"
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/SimpleButton
Multiple libraries were found for "ESP8266mDNS.h"
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/ESP8266mDNS
Multiple libraries were found for "Wire.h"
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/Wire
Multiple libraries were found for "SPI.h"
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/SPI
Multiple libraries were found for "ESP8266WiFi.h"
 Used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/ESP8266WiFi
exit status 1
no matching function for call to 'ArduinoJson::DynamicJsonBuffer::DynamicJsonBuffer(int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

</details>
<br><br><br>

tobozo, 2019-11-18:
Used: /home/ninja/Arduino/libraries/ArduinoJson
   Not used: /home/ninja/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/ArduinoJson

delete the "Used" folder (`/home/ninja/Arduino/libraries/ArduinoJson`) and it'll use the other one from the deauther sdk

the-mcnaveen, 2019-11-18:
Thanks man, It worked.

closed 2019-11-18 by the-mcnaveen

Ryan-techie, 2023-01-17:
bro, what about Mac ?