SpacehuhnTech/esp8266_deauther #1182: Problem uploading code to the ESP8266 ! (ArduinoJson.h Missing.)

cristi091, 2019-11-27:
<details>
  <summary> Compile log (...)</summary>

  

</details>
<br><br><br>

altimmons, 2019-12-07:
So fix your libraries...

Try using platform io in vs code. 

In your platform.ini file specify each library under lib_deps=

itsmemohit, 2019-12-31:
- Delete all libraries named EEPROM.h
Download library from below given link: 
https://github.com/PaulStoffregen/EEPROMand locate the library from this link

Multiple libraries were found for "ArduinoJson.h"
SIMILARLY Delete libraries naming ArduinoJson.h and download the lobrary from here :https://github.com/bblanchon/ArduinoJson
 and locate in your system

jLynx, 2020-02-24:
https://github.com/spacehuhn/esp8266_deauther/wiki

closed 2020-02-24 by jLynx