SpacehuhnTech/esp8266_deauther #1183: Uploaded deauther using arduino to watch

chaserit123, 2019-11-27:
I used the arsuino project and uploaded to my deauther watch and now the display no longer works?

beckerichmika, 2019-12-16:
Did you select the right board type? Read carefully point 3 on here:https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide to make sure you select the deauth one

wei2374, 2019-12-18:
I have a similar problem, I have configured everything as the tutorial said, also the config.h file is already changed for wristband, but still after a successful flash there is no display on the screen.

gorkalombar, 2020-01-13:
Same happend to me! I bought it like 1 month ago, and as soon as I opened it an switched on, leds are working but nothing is shown on display! I tried flashing 3 different .bin files with the N2D, but none of them is working.
Please, help!!!

DangerousPony204, 2020-06-13:
just go into your boards network 'pwned' and check enable display again under settings, works for me everytime

closed 2020-09-22 by stale[bot]