SpacehuhnTech/esp8266_deauther #1185: Deauther stops working after plugged in a laptop

emrrar, 2019-12-01:
Several months ago I purchased a **WiFi USB Deauther** from here (which is refering to the spacehuhn deauther project): _https://maltronics.com/collections/deauthers/products/wifi-deauther-usb_.
For the past few months it was working just fine (I could scan and deauth with no problem).
I was doing all of these on my phone with the USB Deauther plugged in on my router.

I bought a new laptop a week ago and wanted to try the deauther on the same laptop.
I plugged the USB in and when trying to SCAN APS it didn't do anything.
Sometimes it just throws error, sometimes just stands by.
The moment i plugged this USB into my laptop it just stopped the SCAN APS function no matter if I'm using it from my phone or laptop.
And the **problem** is that this **deauther won't even work no matter where it is plugged in now**.

I thought installing the N2D would fix the problem so I downloaded it from here: _https://github.com/mrvodka007/n2d_.
It didn't fix the problem.
Everything I searched for didn't help me.
The USB Deauther doesn't even flash green now. It only has some tiny blue light in the back of the device. It used to switch from bigger blue and green lights. But not now.

So please could someone tell me why is that and if we could fix the problem?

beckerichmika, 2019-12-16:
Check with a multimeter if you're getting stable voltages on the power pins. Either your esp is fried or you are in a bootloader mode or it doesn't get enough power. What does lsusb show (or what driver does windows associate to it)?

emrrar, 2019-12-16:
I'm pretty sure it is fried by now. I tried it to plug it on other devices but the results are the same.
So I'm almost certain that it won't work ever again (the SCAN APs )

beckerichmika, 2019-12-16:
So reflashing doesn't work? I bet there are sketches somewhere on here to check the wifi capabilities or you could write a simple sketch that outputs the networks to the console. Maybe you can give that a try

emrrar, 2019-12-16:
Yes, I did try to reflash but it didn't do anything so I've given up hope.
However I'm thinking since it is "burned" inside nothing will fix it unless I buy something similar (like the OLED one).
I'm thinking of buying one of those OLED deauthers and powering it up through a micro USB on that same laptop again. My one concern is: Can the same issue happen again when powering the OLED deauther with my laptop through a micro USB? Can it "fry" again or something like that?
I would be grateful if you give me an answer.

beckerichmika, 2019-12-16:
Normally it shouldn't happen unless your laptop and your deauther are lying on a conductive surface. You should also check how much current your port is outputting or maybe use a phone charger and power the device with it instead. Afaik the power regulators of the deauther should be able to prevent such accidents. I don't know if there is a fuse on the deauther but you should check if you still have some hope for your deauther.

closed 2020-09-22 by stale[bot]