SpacehuhnTech/esp8266_deauther #1186: 192.168.4.1 cannot load

fengshuo2004, 2019-12-01:
- [x] Flashed .bin file into ESP8266 sucessfully using **ESP8266Flasher**
- [x] Serial interface works perfectly and is fully functional
- [x] WiFi access point is connectable

However I am unable to use the web interface by visiting `192.168.4.1`.

My browser (Firefox) just says "Unable to establish connection to server 192.168.4.1", which is rather unintuitive.

Same thing happens when trying to access _deauth.me_, as it just resolves to `192.168.4.1`.

However, pinging 192.168.4.1 using the command line like this:

```
ping 192.168.4.1
```

does give good responses:

> 正在 Ping 192.168.4.1 具有 32 字节的数据:
> 来自 192.168.4.1 的回复: 字节=32 时间=2ms TTL=255
> 来自 192.168.4.1 的回复: 字节=32 时间=3ms TTL=255
> 来自 192.168.4.1 的回复: 字节=32 时间=3ms TTL=255
> 来自 192.168.4.1 的回复: 字节=32 时间=4ms TTL=255
> 
> 192.168.4.1 的 Ping 统计信息:
>     数据包: 已发送 = 4，已接收 = 4，丢失 = 0 (0% 丢失)，
> 往返行程的估计时间(以毫秒为单位):
>     最短 = 2ms，最长 = 4ms，平均 = 3ms

Additionally, no errors are thrown onto the serial interface as this happens.

<details>
  <summary> Compile log (...)</summary>

  N/A - flashed with **deauther_2.1.0_4mb.bin**

</details>
<br><br><br>

TheMiguelBi, 2019-12-11:
This happens to me too. Wemos D1 mini. 
Have you found the solution already?

TheMiguelBi, 2019-12-11:
My serial monitor is this:

-------------
 1264, room 16 
tail 0
chksum sd l⸮⸮| ⸮$⸮|⸮d⸮c|⸮⸮⸮⸮s⸮c⸮c⸮⸮go⸮$go⸮⸮⸮b8⸮⸮ds$s$p⸮n⸮⸮l⸮⸮#n⸮|l⸮⸮⸮⸮#⸮⸮o'⸮ d⸮⸮l ⸮go$`or⸮ۓo⸮⸮d`r⸮⸮n⸮⸮l`⸮⸮o⸮
Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253cf1 depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff4a98 0000000f 0000000b 00000007  
3fff1c80:  40253cf1 0000013e 3fff4ad8 0000000f  
3fff1c90:  00000002 3fff4af8 3fff0778 402093db  
3fff1ca0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1cb0:  3fff0918 3fff4bdc 00000000 00000000  
3fff1cc0:  00000000 3fff093c 3fff0c28 40266a31  
3fff1cd0:  3fff1d44 3fff0c28 3fff093c 4021f72f  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cec  
3fff1d00:  3fffdc20 00000000 3fff0ce5 4022616e  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253cf1 depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff4a98 0000000f 0000000b 00000007  
3fff1c80:  40253cf1 0000013e 3fff4ad8 0000000f  
3fff1c90:  00000002 3fff4af8 3fff0778 402093db  
3fff1ca0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1cb0:  3fff0918 3fff4bdc 00000000 00000000  
3fff1cc0:  00000000 3fff093c 3fff0c28 40266a31  
3fff1cd0:  3fff1d44 3fff0c28 3fff093c 4021f72f  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cec  
3fff1d00:  3fffdc20 00000000 3fff0ce5 4022616e  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

[... 276 lines cut ...]
3fff1ca0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1cb0:  3fff0918 3fff4cfc 00000000 00000000  
3fff1cc0:  00000000 3fff093c 3fff0c28 40266a31  
3fff1cd0:  3fff1d44 3fff0c28 3fff093c 4021f72f  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cec  
3fff1d00:  3fffdc20 00000000 3fff0ce5 4022616e  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...

Infinity Loop
-------------

But the pwned network is not found!

GunnerBasil, 2020-01-15:
try connecting to pwned, then going to deauth.me
it's another way to get to the site. :D

gorkalombar, 2020-01-18:
Make sure you are not connected to a another second internet access. I mean: if you are trying to access from Android, connect to the pwned WiFi, and **turn off the phone data**. If the phone is connected to the WiFi and to the 4G, and it can't access internet by wifi, it will ask the request for the website "192.168.4.1" from the 4G.
Make sure you are only connected to the wifi.

GunnerBasil, 2020-01-18:
I'm trying to connect on my chromebook.

Kirubakaran786, 2020-02-29:
192.168.4.1 is not open's. My device is only connected to pwned also.

closed 2020-09-22 by stale[bot]