SpacehuhnTech/esp8266_deauther #1187: esp32?

0x77dev, 2019-12-02:
Maybe some esp32 support?

orgixmh, 2019-12-10:
It is not possible yet, esp32 wifi library rejects deauth packets with error:  "wifi: unsupport frame type: 0a0".

cynay, 2020-02-21:
Should be possible now with this: https://github.com/Jeija/esp32-80211-tx

jLynx, 2020-02-24:
https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#esp32-and-esp8285-support

closed 2020-06-16 by spacehuhn

spacehuhn, 2021-12-12:
Hi! There are many issues on this repo about ESP32 support, so I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want. I appreciate your understanding.