SpacehuhnTech/esp8266_deauther #1189: Nodemcu bricked after flashing esp8266_deauther

Raidr1212, 2019-12-11:
I have flashed the latest version of the deauther and it works perfectly. The only problem is my computer no longer recognizes the nodemcu 
[image] I am unable to flash anything else.

itsmemohit, 2019-12-31:
Connect your node mcu which is not being recognized by Windows. You will see an Unknown USB device ( Device Descriptor Request Failed) with the yellow sign in Universal Serial Bus controllers. Now right-click on it and click Uninstall to remove them. Restart your PC and the drivers will be automatically installed
it  worked for me

Raidr1212, 2019-12-31:
> Connect your node mcu which is not being recognized by Windows. You will see an Unknown USB device ( Device Descriptor Request Failed) with the yellow sign in Universal Serial Bus controllers. Now [...]

That sadly has not worked for me

closed 2020-12-23 by stale[bot]