SpacehuhnTech/esp8266_deauther #1190: arduino-esp8266-deauther removed from arduino application

lesquieshia, 2019-12-12:
I don't have any issues with coding, I am following the installation guide at "https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#flashing-the-firmware-bin-file" and there seems to be a step that I can't do. The guide essentially says to download arduino-esp8266-deauther in arduino after going to tools, manage boards, and searching eps8266, but nothing comes up, only the first one it says to download, called "esp8266 by ESP8266 community"

Help?
[image]

Duffman2, 2019-12-19:
I am having the same issue

tobozo, 2019-12-19:
you probably skipped [this step](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide)

Duffman2, 2019-12-19:
I have installed and selected the board that says modified for ARM support, because that is the only one that shows up. I also have the additional boards links set up in preferences. I am on windows. Is there anything else that might have gone wrong? Thx

tobozo, 2019-12-19:
If you managed to install and select the board that says modified for ARM support, then you don't have the same issue.

You'll probably need to provide more details on what you expect vs what you get.

CareerCargo, 2019-12-20:
Hello, thanks for your work - I have the same issue.  esp8266 community and tobozo's 8266-deauther aren't available even with the 2 board manager URLs.  I checked on another one of my computers that had this installed already and they're not coming up in the board manager as installed, even though they were installed a month ago.
Something has certainly changed.  thanks again, hope we can find a solution.... Merry Christmas.

tobozo, 2019-12-20:
@CareerCargo what does the log window display when you open the boards manager?

First you should make sure you're using the new github URL to the [json package](https://raw.githubusercontent.com/wiki/tobozo/Arduino/package_deauther_index.json) since this has recently changed.

Obviously the Arduino IDE must be allowed to access the network in order to get those boards listed.

When opening the boards manager for the first time, the log window will probably show a signature error (package.json does not have a .sig file), but it will work fine on second attempt.

If you need more help on this, please provide more details (OS, IDE version, LAN filtering conditions) on your situation.

CareerCargo, 2019-12-20:
Thanks for that.  running as admin, Win10, I've went back to 1.8.9 from 10 version to test that, no LAN filtering.
Boards manager comes up normally (as expected - no errors), but it does seem to take longer to grab the data list libraries.
Checked both URL's at least 3 times - all good.
I tried to manually install the Zip library from
https://github.com/tobozo/Arduino/archive/deauther.zip
As per Arduino's explanation via 'sketch' 'include libraries' 'add ZIP library'.
Error states, - Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno" Specified folder/zip file does not contain a valid library.  Maybe I'm doing this wrong...?
Is there another way to manually install the libraries.

JMC1973, 2019-12-31:
Hi Career Cargo, you can download the library zip file and unzip in user/Documents/Arduino/Libraries
Then rename the folder with the same name without “ -master “.

tobozo, 2019-12-31:
that's not a library, that's a board package, you need to install it from the boards manager, not from the libraries manager

closed 2020-09-22 by stale[bot]