SpacehuhnTech/esp8266_deauther #1191: Error message when trying to upload in arduino

lesquieshia, 2019-12-12:
'''Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Uno"

esp8266_deauther:12:12: error: user_interface.h: No such file or directory

   #include "user_interface.h"

            ^~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1
user_interface.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.'''

beckerichmika, 2019-12-16:
First install the esp8266-specific board configs and select it as upload target because you selected the Arduino UNO instead of the esp8266 which can lead to such errors. Then try again

jLynx, 2020-02-24:
RTFM https://github.com/spacehuhn/esp8266_deauther/wiki

closed 2020-02-24 by jLynx