SpacehuhnTech/esp8266_deauther #1192: ESP8266 not coming up in wifi settings after flashing, constant blue light

marleyqp, 2019-12-13:
I don't have compile logs as i used NodeMCU flasher. (im a noob so sorry 👍 ) After flashing, I plug my board into a microusb cable and as soon as i plug in the board it turns on a blue light. it never did this before. I cant even find it in Wifi settings. Have no clue whats going on here.

GunnerBasil, 2020-01-07:
same, except i used arduino ide
I wish they actually answered

forseenink3238, 2020-01-08:
Same here, my solution is to press the RST button at least 2-3 times.

GunnerBasil, 2020-01-09:
Ok guys, so lets share what we did and what we know. I used arduino IDE to flash the code onto the esp8266. It just deauthed all wifi and blinked every once in a while. I had to spam rst then unplug fast so it would stop, and it didn't display pwned. you?

forseenink3238, 2020-01-09:
Did you guys use node2deauther?

forseenink3238, 2020-01-09:
@marleyqp have you ever uploaded any other code to the nodemcu before you got to know about the esp8266 deauther?

forseenink3238, 2020-01-09:
@marleyqp if you have uploaded any other code to your nodemcu then I would recommend you to reset and erase all code in your nodemcu and then use the nodemcu flasher, to erase the code you need to download a software named Flash Download Tools. You can get the software on espressif's website.

marleyqp, 2020-01-09:
> Ok guys, so lets share what we did and what we know. I used arduino IDE to flash the code onto the esp8266. It just deauthed all wifi and blinked every once in a while. I had to spam rst then unplug fast so it would stop, and it didn't display pwned. you?

I used Node2Deauther I think it was called

marleyqp, 2020-01-09:
> @marleyqp have you ever uploaded any other code to the nodemcu before you got to know about the esp8266 deauther?

No, not apart from the Deauth code

marleyqp, 2020-01-09:
> @marleyqp if you have uploaded any other code to your nodemcu then I would recommend you to reset and erase all code in your nodemcu and then use the nodemcu flasher, to erase the code you need to download a software named Flash Download Tools. You can get the software on espressif's website.

I have tried this unfortunately with no results. Do you think the board could be damaged?

forseenink3238, 2020-01-09:
It is impossible that the board could be damaged, I only had the issue while using node2deauther

forseenink3238, 2020-01-09:
Is your nodemcu using a Cp2102?

forseenink3238, 2020-01-09:
Try uploading a blink example to your nodemcu using arduino ide. To see if it is working.

GunnerBasil, 2020-01-09:
So it's working for you? I didn't upload anything except for the deauther code, and I usb esptool.py to reset it. I opened the terminal in linux and ran the following commands:
pip install esptool --user
esptool.py erase_flash
is this a problem

marleyqp, 2020-04-05:
Working now.

closed 2020-04-05 by marleyqp