SpacehuhnTech/esp8266_deauther #1193: WemosD1mini + V2.5 = CONTINUOUS LOOP !

TheMiguelBi, 2019-12-17:
Wemos D1 mini
Working with version 1.6 and with version 2.0.5
---
Board: Generic ESP8266 Module  
Flash Mode: DOUT
Flash Frequency: 80 MHZ
CPU Frequency: 160 MHz
Flash Size: 1M (256K SPIFFS)
Reset Method: nodemcu
Upload Speed: 115200
Port: <com port of your device>
---

Load version 2.5 (last) and the card restarts continuously.
Does anyone have the solution ?? Thanks!
On the serial I read this:

<details>
  <summary> Compile log (...)</summary>

 1264, room 16 
tail 0
chksum {l l⸮⸮| ⸮d⸮|⸮$⸮c|⸮⸮⸮⸮;⸮#⸮c⸮⸮go⸮do'⸮⸮⸮cp⸮⸮ds$s$p⸮'⸮⸮d⸮⸮#g⸮|d⸮⸮⸮⸮b⸮⸮o'⸮ lćd`⸮'o$`'{⸮⸮⸮g⸮⸮d`{⸮⸮o⸮⸮l`⸮⸮'⸮
Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253cf1 depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff27c0 0000000f 0000000b 00000007  
3fff1c80:  40253cf1 0000013e 3fff2800 0000000f  
3fff1c90:  00000002 3fff2820 3fff0778 402093db  
3fff1ca0:  3fff27e0 0000000f 00000001 3fff26e8  
3fff1cb0:  3fff0918 3fff5024 00000000 00000000  
3fff1cc0:  00000000 3fff093c 3fff0c28 40266a31  
3fff1cd0:  3fff1d44 3fff0c28 3fff093c 4021f72f  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cec  
3fff1d00:  3fffdc20 00000000 3fff0ce5 4022616e  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253cf1 depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff4a98 0000000f 0000000b 00000007  
3fff1c80:  40253cf1 0000013e 3fff4ad8 0000000f  
3fff1c90:  00000002 3fff4af8 3fff0778 402093db  
3fff1ca0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1cb0:  3fff0918 3fff4bdc 00000000 00000000  
3fff1cc0:  00000000 3fff093c 3fff0c28 40266a31  
3fff1cd0:  3fff1d44 3fff0c28 3fff093c 4021f72f  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cec  
3fff1d00:  3fffdc20 00000000 3fff0ce5 4022616e  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40253cf1 depc=0x00000000

ctx: cont 
sp: 3fff1ad0 end: 3fff1d20 offset: 01a0

>>>stack>>>
3fff1c70:  3fff4a98 0000000f 0000000b 00000007  
3fff1c80:  40253cf1 0000013e 3fff4ad8 0000000f  
3fff1c90:  00000002 3fff4af8 3fff0778 402093db  
3fff1ca0:  3fff4ab8 0000000f 00000001 3fff2aa0  
3fff1cb0:  3fff0918 3fff4bdc 00000000 00000000  
3fff1cc0:  00000000 3fff093c 3fff0c28 40266a31  
3fff1cd0:  3fff1d44 3fff0c28 3fff093c 4021f72f  
3fff1ce0:  00000000 00000000 00000000 00000000  
3fff1cf0:  00000000 00000000 00000000 3fff0cec  
3fff1d00:  3fffdc20 00000000 3fff0ce5 4022616e  
3fff1d10:  00000000 00000000 3fff0d00 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

</details>
<br><br><br>

tobozo, 2019-12-17:
sounds like a corrupted json file on the spiffs partition, maybe re-apply the spiffs uploader ?

also checking [similar issues](https://github.com/spacehuhn/esp8266_deauther/search?q=boot+mode%3A%283%2C7%29&type=Issues) may help you in find the reason of this boot loop, and hopefully fix it too

TheMiguelBi, 2019-12-17:
Thanks! (to be honest I didn't even know what to look for!)
I found similar problems. 
Now I try some roads..

Lisheng2016, 2020-05-24:
hey, just went into exactly the same problem,but it works with precomiled binary file  deauther_2.1.0_DSTIKE_OLED_v2_v2.5_v3.bin
i'm using SH1106 1.3'' OLED,as stated in A_config.h,DSTIKE_OLED_v2_v2.5_v3 uses the same hardware setup,so i flashed it with bin file for DSTIKE_OLED_v2_v2.5_v3 and it worked

bking46, 2020-05-29:
@Lisheng2016, thanks for sharing this. Before I go flash the bin file, is this the board you used?
With this board I had the same issue with the latest deauther version when compiling and uploading via Arduino. Only got it to work with the older version of deauther v1.6. Please let me know if you have the same board as me, so I can check if flashing the precompiled bin would work with the latest version. Many thanks
[image]

Lisheng2016, 2020-05-31:
> @Lisheng2016, thanks for sharing this. Before I go flash the bin file, is this the board you used?
> With this board I had the same issue with the latest deauther version when compiling and uploading [...]

I'm using D1 mini pro.The new precomplied bin file "DSTIKE Deauther OLED v2, v2.5, v3" or 
"DSTIKE Deauther OLED v3.5, v4" would work for your board/display,but may not work with the buttons used for select/up/down,take a look at the config file of the precomplied binary and compare it to your hardware setup,if they're the same,then it should work.

closed 2020-09-22 by stale[bot]