SpacehuhnTech/esp8266_deauther #1194: stop attack from serial input

dave0111, 2019-12-17:
i wanted to know if there is any way of stopping an attack by sending a serial command other than attack timeout
<details>
  <summary> Compile log (...)</summary>

  

</details>
<br><br><br>

unguest, 2020-05-10:
You can connect to a GPIO/Serial port (it depends what device you are using) and then code that yourself. It should looks like something like that : 

```C
#define atkPort 8 // Just taking arbitrary values in the whole program, adapt to your needs

void setup() {
      [your setup function here]
      pinMode(atkPort, INPUT);
}

void loop() {
     [your loop function here]
     if(digitalRead(atkPort) == LOW) {
          // TODO : stop attack
     }
}
```

Ozymandias1700, 2020-05-24:
If you are controlling via serial UART and sending commands, why can't you simply type `stop`?

closed 2020-09-22 by stale[bot]