SpacehuhnTech/esp8266_deauther #1202: Can't connect to pwned on Chromebook

GunnerBasil, 2020-01-13:
I tried to connect to pwned on my chromebook, it doesn't allow me to connect. When I try on my macbook, it works and I can deauth. I think this is because pwned doesn't have an active network connection, so it doesn't let me connect. what is a workaround?

unguest, 2020-05-10:
Connect to the 'pwned' wifi on your phone and then share the connection of your phone via an USB cable. Then go to the phone's settings and you should be able to look at the deauther IP in the wifi settings.

spacehuhn, 2020-06-17:
Maybe the chromebook won't let you stay connected, because the network doesn't provide an internet connection. Surely there must be some sort of setting to make chromebooks connect to these networks anyway.

unguest, 2020-06-17:
@GunnerBasil are you sure your network & security settings are complying with the use of non-trusted networks ?

closed 2020-09-22 by stale[bot]