SpacehuhnTech/esp8266_deauther #1206: DStike Wristband V2 Switching down every 32 seconds

gorkalombar, 2020-01-18:
Recently I bought the Dstike watch V2 (with the RTC). I have tried flashing the code many times, changing many things, in order to check what was generating my problem, but I can't find the origin.
The problem is that everytime I flash de code, the watch works perfectly for a couple of times (I mean "boot"-s), but after a while, everytime I turn it on, it goes off after 32-34 seconds. Always between 32 and 34 seconds.
One detail I noticed, is that I can know when the watch is going to work well, or automatically switch down, because the LEDs which indicate the battery I have left, are keepig on if the watch is going to stay on. If the watch is going to turn off after that 32-34 seconds, the battery lights are turning off after 10 seconds of booting up.

I plugged the watch to the Arduino monitor, and this is what you can see on screen in the moment of turning off automatically:

```Exception (2):
epc1=0x3fff5d48 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5d48 depc=0x00000000

ctx: cont 
sp: 3fff1be0 end: 3fff1e90 offset: 01a0

>>>stack>>>
3fff1d80:  00000000 3fff1dc0 3fff5d48 40228202  
3fff1d90:  3ffefdac 3fff0728 3fff1dfc 40212a8f  
3fff1da0:  3ffe9694 3fff5d48 3fff1dc0 40212ac8  
3fff1db0:  00000030 00000000 3fff0714 40210935  
3fff1dc0:  3fff5c90 00000000 40210058 4020d1bc  
3fff1dd0:  00000000 00000000 3fff5d48 40211b4c  
3fff1de0:  00000028 3fff0e64 3fff1e20 4022133a  
3fff1df0:  40221c98 3fff5428 00000005 3fff5c70  
3fff1e00:  0000000f 00000000 3fff5778 4022194c  
3fff1e10:  00000010 3ffefdac 3fff6560 4022162b  
3fff1e20:  3fffdc20 3ffefdac 3ffefe04 3fff0e64  
3fff1e30:  3ffefdac 3fff0728 3fff0714 402133ed  
3fff1e40:  3fffdc20 3ffefdac 3fff0714 40213476  
3fff1e50:  3fffdc20 3ffefdac 3fff0e5d 4021dab6  
3fff1e60:  40226475 00000000 00000000 3fff0e64  
3fff1e70:  3fffdc20 00000000 3fff0e5d 4022649d  
3fff1e80:  00000000 00000000 3fff0e70 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```
And it restarts again.

I'm quite new into this, so sorry if this is not the right place for posting. But I don´t know what more to do. Can't find anything in google about this.
Thanks!

gorkalombar, 2020-01-18:
I found something about the rst cause: something related to the "reset pin". In a forum I found someone saying something like "it could indicate power supply issues if nothing has actually happened to the pin itself", but I'm not sure if this is my problem, because the watch came assembled and with the correct battery on it. Then I saw this: "the boot mode for this one is boot from spi flash memory". And I don't know what this means or how to solve it. Maybe something related to a delay into the code...? I don't know.

meditant, 2020-05-24:
I have the same problem, and it's work with #define RESET_SETTINGS in A-config.h

I thinks it's a eeprom config problem.... but i not search more.

gorkalombar, 2020-05-25:
@meditant I still have the same problem... I bought the watch 5 months ago, and I can´t use it becasue of this.
Thanks for your idea. I will investigate more about _eeprom_. 👍

Dgibson80, 2020-08-09:
Hey all, 

I too am experiencing the issue OP has explained.  Still cant figure out whats causing it.  If I turn the watch on for too long the battery indicator vanishes, and will power off like mentioned above (every 30 seconds or so.)  If the watch is plugged in it will stay on. Setting the display timer to zero will just cause the device to boot without the OLED.  Would really like a solution to this problem if anyone can suggest anything to try.  

Thanks

gorkalombar, 2020-08-11:
I gave up many weeks ago trying to find a solution. I like to see that I'm not the only one with this problem, but after 8 months, it's not worth it the research. I have been quite disappointed I have to say.

closed 2020-11-16 by stale[bot]