SpacehuhnTech/esp8266_deauther #1207: Dstike Wristband Oled/SSID Clone/Settings Problem

benb0jangles, 2020-01-18:
I have a problem with my Dstike V2 Wristband in that it has stopped working correctly.
- Oled has mostly stopped working. If I turn off the wristband for several hours, then turn on, the OLED tries to work but then freezes halfway through the bootscreen. After that it does not work.
- I am unable to 'clear all' the list of Cloned SSIDs. After I save and reload, they re appear.
- I am also unable to view settings after connecting to the wristband with either smartphone or laptop

Has this been reported before?

gorkalombar, 2020-01-20:
Hi! I'm having many problems with the Dstike V2 Wristband too; I opened a couple of issues some days ago.
I had problems with the OLED too. When I turned it on fot he first time, the oled screen was not working, and it was not going into flashing mode. The solution finally was to unplug and plug again the battery. I don't know why, but it worked.
I also recommend you to check if all the resistances are properly soldered. One of mine was not, and I had to resolder it again.
Now the watch seems to work ok, but it's turning off itself every 34 seconds. I have tried desesperatly many many things and changes, but anything is working.

Considering its "high" price, and that they are not comming with any warranty, I would not recommend this board, especially if you are a beginner.

spacehuhn, 2020-06-17:
If the OLED is not turning on try connecting to 'pwned' and check enable display under settings

closed 2020-09-22 by stale[bot]