SpacehuhnTech/esp8266_deauther #1208: Deather doesn't work

Rafo124, 2020-01-22:
My ESP8266 doesn't death my network. I tried it on my network but nothing happens. Is my modem using 802.11w-2009? Or is it something bad with the board

jLynx, 2020-02-24:
Are you trying on 5 or 2.4 Ghz?

ricksmith96, 2020-02-24:
This happened to me too it says it death but it doesn't to anything

hoangminh5210119, 2020-02-25:
I have a problem when rebuild your source code, deauther not work but upload bin file work well, i'm use  Arduino IDE and platformio VSCode to build with newest platform version, please help me

jLynx, 2020-02-25:
@ricksmith96 are you trying it on 5 or 2.4 Ghz?

kuno23, 2020-03-15:
same problem here.. i have 2board.. nodemcu and wemos d1 mini pro with oled (project known as MagiCube) .. i can connect to the wifi portal, i see scanned wifi, but when i sellect an wifi (of course my home wifi) and when i select deauth attack,  start the attack, it looks,that it do something, but in real it does nothing, no deauth my devices from wifi. the same is with my wemos r1 mini pro, through the menu on lcd i select Ap, then deauth attack, but nothing.i dont know where can be the problem. In NodeMCU i have this image  deauther_2.1.0_4mb.bin, in Wemos i have image from GitHub repository, only changed the pins for buttons and oled.

jLynx, 2020-03-15:
@kuno23 are you trying it on 5 or 2.4 Ghz?

kuno23, 2020-03-16:
of course on 2,4ghz wifi.. i don´t think that it is possible to do on 5ghz if the  esp8266 has only 2,4ghz.

jLynx, 2020-03-22:
Closing as no one seems to have this problem anymore

closed 2020-03-22 by jLynx