SpacehuhnTech/esp8266_deauther #1209: Every link in Releases for v2.1.0 and v2.0.5 is broken

Rub-z, 2020-01-23:
I've tried to download **any** link from Releases for v2.1.0 and v2.0.5 and I get a server connection error.

For example clicking on the 'deauther_2.1.0_1mb.bin' link results in the following error:

Firefox can’t establish a connection to the server at github-production-release-asset-2e65be.s3.amazonaws.com.

The URL it tries to connect to:

https://github-production-release-asset-2e65be.s3.amazonaws.com/78281075/4c3ba680-efef-11e8-8e5c-4f7df73e6078?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200123%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200123T111040Z&X-Amz-Expires=300&X-Amz-Signature=8da2cc74c5ae048106e5aba3aecd2fdf7a530fd0d185115abc6f650d3f9ca242&X-Amz-SignedHeaders=host&actor_id=46804304&response-content-disposition=attachment%3B%20filename%3Ddeauther_2.1.0_1mb.bin&response-content-type=application%2Foctet-stream

Every link in Releases does the same thing.  I've tried 3 different browsers too.

Perhaps there is a mirror?  Any help much appreciated.

Cheers

tobozo, 2020-01-23:
Please don't paste links with credentials, not sure what it's exposing here but just imagine these could be related to your account :-)

Were you using a VPN when the problem occured ? Maybe choose a different exit country.

Rub-z, 2020-01-23:
>Please don't paste links with credentials, not sure what it's exposing here but just imagine these could be related to your account :-)

That appears to be the link whether logged in or not.  I've removed the credential from the orignal link just in case.

>Were you using a VPN when the problem occured ? Maybe choose a different exit country.

no VPN.  **Every other** github project loads no issues and I can download any release.

Are you able to download?

[deauther_2.1.0_1mb.bin](https://github.com/spacehuhn/esp8266_deauther/releases/download/v2.1.0/deauther_2.1.0_1mb.bin) from the releases page?

tobozo, 2020-01-23:
yep I tested a few links and while there is a redirection to AWS, it works just fine from my location (Paris, France)

If it isn't related to your location, maybe it is related to your LAN settings, would you have a PiHole or something that could interfer with your DNS ?

Rub-z, 2020-01-23:
merci beaucoup! yes, turns out the network has a pi-hole. I got the admin to tmporarily disable it and I was able to download. Thanks for your help and suggestions :-) . I'll close the issue.

closed 2020-01-23 by Rub-z

tobozo, 2020-01-23:
very curious to know what rule on the PiHole blocks this repository, can your admin get you this information ?

Rub-z, 2020-01-24:
admin is away until next week but i'll ask when they get back ;-)

Rub-z, 2020-02-01:
finally have an answer for you.  A while a go an Amazon s3 bucket was serving malware so my Admin blocked the DNS (s3-1-w.amazonaws.com) and IP (52.216.131.107).  The server is no longer compromised and is now whitelisted.  So, it was a custom rule which caused the issue and now everything is working correctly.  Thanks for your help.