SpacehuhnTech/esp8266_deauther #1212: Can't open 192.168.4.1

crone11, 2020-02-03:
I bought the D-Strike USB De-Auther v1 and I can see and connect to the pwned Wi-Fi network. Problem is, after that I can't connect to 192.168.1.4 because Firefox says `The connection has timed out | The server at 192.168.4.1 is taking too long to respond.` I get IP from DHCP and everything. I have tried using a different browser, different machine, phone and re-flashing the device via the N2D2 application.

I also can't connect to it via the serial in the Arduino IDE. After going to Serial Monitor and setting the appropriate options (Newline, 115200 baud) nothing gets output, even after entering the command `sysinfo`.

richrd, 2020-02-03:
Duplicate of #1186

crone11, 2020-02-03:
@richrd Not completely, as for me the serial interface is not working either.

crone11, 2020-02-03:
Additional info: DNS server also works (`nslookup deauth.me` returns `192.168.4.1`)

crone11, 2020-02-06:
I don't know what the cause of the original problem was and/or how to solve it, but I've managed to get this working for myself.

When I was flashing the board, it didn't actually flash it, as the COM1 port was my motherboard's serial port, and not the USB to UART adpater. This was also why the serial port was not working for me. The provided auto flashing tool said the flashing was done, but it actually wasn't, it just throw the output to my motherboard's serial port. I had to install the necessary drivers, and after compiling & lashing the board via the Arduino IDE, it now works.

closed 2020-02-06 by crone11