SpacehuhnTech/esp8266_deauther #1213: Missing smd-part and bad soldered part on pcb

stefanl110, 2020-02-05:
Hi there, today my esp3286 arrived and i got shocked when i looked on the pcb board.
There is a missing part on the bord, i think its a resistor and above there is a bad soldered part. What can i do now??
Should i send it back or is there a soldering schematic where i can identify the missing part and solder the other one in the right place?
Hope that somebody can help me. 
See my pics:

[image]

Thats the way it should be
[image]

Thx for the help

tobozo, 2020-02-05:
have you tried to solder it back where it should be ? at best this seems to only power a LED and at worst it's used for debouncing the three ways button

stefanl110, 2020-02-06:
Sure i could solder the transistor in the right position, but the missing resistor is not included in the bag. I dont know which resistor it should be.
If i do the soldering, i will lose the warranty.

tobozo, 2020-02-06:
can you add precision to your post and specify what model it is ? I can't find it on [dstike repo](https://github.com/lspoplove)

speculation: resistor could be a 47K or 10K given its proximity to the three-ways push button

stefanl110, 2020-02-06:
Its the „esp8266 deauther“, i dont know the serial number yet. I am not at home. 
i didnt bought it from dstike repo, i bought it on amazon. 
MakerHawk ESP8266 WiFi Deauther... https://www.amazon.de/dp/B07HC1RDTM?ref=ppx_pop_mob_ap_share

tobozo, 2020-02-06:
well it's clearly between the transistor and a LED, so you can safely start with a 47K resistor and use a 10K if the led stays off, all in all it does not prevent the deauther to work as long as you don't use a battery, that's still something to play with while you wait for a replacement unit

[image]

stefanl110, 2020-02-06:
Yes exactly under the transistor.
I am sure i could test this, but i will lose my warranty if i solder on it or start trying which of the resistor will fit. 
Amazon would like me to send it back to the seller, if i do so i couldnt try it.
But the seller answered my mail about this faulty pcb-board. I sent some pics and now i will wait until they answer again. 
But thx for the information, maybe i will need this info. We will see

stefanl110, 2020-02-08:
So, the seller send me a new one, without sending the old one back...nice.
So i could solder a resistor on it and solder the transister on the right place and than i have two. Very nice...
So thank you for the information about the resistor. Thx all for the help.

tobozo, 2020-02-08:
- [x] purchases one unit
- [x] get the smd experience 
- [x] get an extra unit

I wish it went that well with every purchase ^^

closing this issue as resolved

closed 2020-02-08 by tobozo

stefanl110, 2020-02-08:
> * [x]  purchases one unit
> * [x]  get the smd experience
> * [x]  get an extra unit
> 
> I wish it went that well with every purchase ^^
> 
> closing this issue as resolved

Oh yes thats true,...i wish that too.