SpacehuhnTech/esp8266_deauther #1214: v2.1.4-easter , how esp8266 - esp32 work?

Neodim7, 2020-02-16:
hello

in  release v2.1.4-easter i see this - ESP32 Support : wire the rx/tx of your ESP8266_deauther to the rx/tx of an ESP32 and enjoy the benefits of serial tunneling!

but what firmware i must will flash to esp32?

thank you

tobozo, 2020-02-16:
you're a month and half early, please come back on April 1st :smile:

Neodim7, 2020-02-17:
i connected it and tryed )) and ... nothing happen... ok i will try it at 1 april ))

tobozo, 2020-02-17:
Actually that's more a way to build your own deauther than an easter egg.

The joke is about the uselessness of doing it, why would you do serial tunneling in the first place?

See [this project](https://github.com/tobozo/M5Stack-ESP12-Shield) as an example of how useless it can be :-)

Neodim7, 2020-02-18:
Mmm, maybe for better heat dissipation :)

closed 2020-06-17 by spacehuhn