SpacehuhnTech/esp8266_deauther #1217: Not able to flash

PineMaster, 2020-02-21:
So i brought this https://www.ebay.com/itm/New-NodeMCU-Lua-ESP8266-ESP-12E-cp2102-Arduino-compatible-wifi-development-board/382593145644?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
recently, and it seems i cant flash the bin file onto it, its a node mcu esp-12E, i also tried using flash mode, connected G1 to gnd and nothin, i tried the node mcu flasher, arduino etc but still didnt work
can anyone help me?

BB3333, 2020-02-27:
Hello , same problem ... i use a super standard Hiletgo NodeMCU CP2102 ESP-12E (AMICA)
and nothing happend. I flash it with ESP8266Flasher , it flash but the device is not having any wifi
What's the difference between deauther_2,1,0_4 MB .bin and the 1 MB bin 
Which one do i need to use ?
This is clearly not obvious
Thanks for your help

BB3333, 2020-02-27:
ok i succeed with the Node2Deauther V2 ;)

closed 2020-03-22 by jLynx