SpacehuhnTech/esp8266_deauther #1221: Device automatically powers Off after 30 Seconds

User00700, 2020-03-01:
Hello Community,
i have an little problem with my deauther. It worked fine for a day but now IT automatically powers Off after 30 seconds. In Detail: i Power on using the Power switch (in batterie Mode with a fully charged accu), after 8 Seconds the 4 accu Power LEDs power Off and if i do Not push the power Switch then after 22 Seconds the device Powers Off. If i Push the Power Button while device is on and the 4 Power LEDs are Off then the power leds switch on again. This Looks a Kind of Power saving or Bug to me. Has anybody an idea for a solution? Interesting fact that there is No issue when i use power via Micro USB Port....

Daniel-strg, 2020-03-13:
I have the same Problem with my device with Micro USB Kabel its works but with Batterie its Turn off like 30 seconds.
I Flash the device New, but the Problem is Not away. 
I have a wifi deauther v5 oled

unguest, 2020-05-10:
Did you check all the electrical parts that are contributing to the circuit power supplying ? (I recommend you to have a look at capacitors / inductances behaviors)

spacehuhn, 2020-06-17:
This is most likely the powerbank cutting off power because the board isn't taking enough energy. The powerbank thinks nothing is charging and turns off to save power after a while.

closed 2020-09-22 by stale[bot]