SpacehuhnTech/esp8266_deauther #1222: Just thanks, close this

Garfius, 2020-03-06:
Thanks for the chicken.

And the whole rest bwt 😜

closed 2020-03-06 by Garfius

richrd, 2020-03-07:
I think this is a duplicate of #1167 or maybe #720  , but I also have this issue. I tried to dearth my chicken but it disappeared and now I cant find it anymore. :chicken: :grey_question:   Please reopen issue as it isn't resolved (and add labels). It works though. Thanks for the great project!

Phoenix1747, 2020-03-07:
So many people trying to earth their chicken nowadays... Should have just used an ESD strap...

richrd, 2020-03-07:
I don't have an ESP Strap, only the ESP 8266. Where can I get an ESP Strap? I haven't heard of that model before.  Also I'm not trying to earth it since I know that the earth is actually flat so it wouldn't work :globe_with_meridians: . Next time I will use the srtap. Anyway eventually I found my chicken, hiding behind the 3/4G modem that I tried to duerth. Still not working but I guess that's expected since the chicken was connected via LAN cable (#1046). I followed the instructions in #716 but my ting not work halp asap plz :cow:  wow

Phoenix1747, 2020-03-07:
You need at least a 100W 5G modem to be able to not lose your :chicken: again, don't forget that.

Also yes, you can drill the hole, but I'd recommend you just use it as a wireless strap. Makes it easier to move around when looking for any disobedient chickens. Something like this will work fine: https://www.ebay.com/itm/Wireless-Anti-Static-Bracelet-ESD-Wrist-Strap-Band-Prevent-Electrostatic-Belt/312991146095