SpacehuhnTech/esp8266_deauther #1226: victim AP keeps changing channels - possible workaround

v0rtex77, 2020-03-22:
Hello, I know that if the victim AP changes channel, deauther has to manually scan again and select the new AP-chan. I have also read that if you have selected multiple APs/channels deauther will hop between. 
Now, would it make sense to have an option to lock on an AP's MAC/SSID and attack by default on all channels ? Make instances of the AP in channels 1-13 and hop ? 
cheers

spacehuhn, 2020-06-17:
[Version 3](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v3) (still in development) allows you to select multiple channels for an attack.

anwareset, 2020-07-10:
> 
> 
> [Version 3](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v3) (still in development) allows you to select multiple channels for an attack.

Automatic channel change detection that locked to SSID/MAC gonna be pretty cool to be implemented.
In case of the victim is try to restart the router and changing channels. keep attacking with the new channel would be an interesting function.