SpacehuhnTech/esp8266_deauther #1231: Recently purchased a Deauther Wristband

Cyne-wulf, 2020-04-18:
This was my first exposure to Arduino, so I know next to nothing about the platform.

Upon starting the watch I found that it wasn't preflashed with the deauthing software. I decided to follow the link printed on the card in the box and it brought me here. I followed three different flashing tutorials unsucessfully:

I first attempted to flash using the Node 2 Deauther GUI program, the program failed to detect the device's MAC and AP MAC values and yes, I did select the right COM port. After troubleshooting that specific program for a while I decided to try the second method: compiling and uploading to the board using the Arduino IDE. I was able to set everything up in the IDE successfully and launched an upload. The compiler output said it was unable to write to the board's memory and that numerous directories "did not exist."  The operation returned unsuccessful and rendered the screen and LED on the front of the watch completely blank (off). Luckily there is a small green LED on the back of the watch that still functioned though. 

In a last ditch attempt to fix the problem I booted into linux, thinking that it might be an issue with Windows. I downloaded the Esptool.py CLI tool using pip and the binary for the watch. The flash command executed and even returned that the operation was successful, however the screen and LED still remain off and I have no evidence that it actually was successful. 

Any steps I can try from this point? I'm willing to give anything a shot as long as it doesn't require purchasing any new hardware! Thanks!

Ja1go, 2020-05-17:
Having the same issue. will post if I find a fix.

Ja1go, 2020-05-17:
Alright, Not 100% fixed but getting closer. This is also my first Arduino so bear with me.  When Performing a new "flash" the display is auto set to Disabled. If you go to the A_config.h tab (second tab in Arduino) and scroll to ==Display CONFIG==, you'll see right below it #define USE_DISPLAY false. Set that to True. Then go further down and #define Dauther_DISPLAY SH1106Wire= ect... for 1.3"OLED. The way you define it is by removing // in front of it and adding // to the display SSD1306Wire above it. you'll see it change colors. Now you'll notice the switch to navigate is not bound correctly. I am working on how to fix that.

Cyne-wulf, 2020-08-08:
Thanks for the suggestion! I'll try out the steps as soon as I have the time and let you know how it turns out.

closed 2020-11-13 by stale[bot]