SpacehuhnTech/esp8266_deauther #1232: selcting the ap and start attack not working.

Alpha-1729, 2020-04-19:
i select the target ap.
and click on the start attact.
but all device can connect to the ap.

richrd, 2020-04-19:
Are the devices connecting on the 2Ghz or 5Ghz band? If they're on 5Ghz it wont work because 5Ghz isn't supported on the ESP 8266.

Alpha-1729, 2020-04-19:
i tried it using my laptop as ap and tablet as client..
not working

richrd, 2020-04-19:
Did you check which band they're on?

Alpha-1729, 2020-04-26:
i didnt't know what you mean.

richrd, 2020-04-26:
This explains what the bands are https://www.howtogeek.com/222249/whats-the-difference-between-2.4-ghz-and-5-ghz-wi-fi-and-which-should-you-use/

Alpha-1729, 2020-04-26:
they are on 2.5ghz

spacehuhn, 2020-06-17:
Have you tried it on other networks?

Alpha-1729, 2020-06-17:
> Have you tried it on other networks?

yes , but client device can connect to the AP

closed 2020-09-22 by stale[bot]