SpacehuhnTech/esp8266_deauther #1233: Unable to update vendor list

vuongfcc, 2020-04-21:
when I perform the vendor list update. I get this error. How to fix it?
`Traceback (most recent call last):
  File "update_manuf.py", line 144, in <module>
    generate_lists(options.url, options.output, options.small)
  File "update_manuf.py", line 39, in generate_lists
    data = urlopen(WS_MANUF_FILE_URL)
  File "/usr/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 563, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 755, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 569, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 403: Forbidden

`

closed 2020-05-14 by vuongfcc

spacehuhn, 2020-05-27:
Were you able to fix the problem?

vuongfcc, 2020-06-06:
I do not know how to fix it. But the error no longer occurs.