SpacehuhnTech/esp8266_deauther #1234: State of the Captive Portal

ydixken, 2020-05-01:
Hi there, 

my DSTIKE (Deauther Monster v4) board just arrived a couple of days ago and I've been fiddling around with it a lot - great project so far! 

In #1143 somebody asked about the captive portal and I think that there might be a misunderstanding. In `settings.h` there a following booleans available:

`bool captivePortal = true; ` & `bool webInterface     = true;`
I've been expecting that a client who connects to a spoofed unencrypted SSID gets routed to a Captive Portal, not to the Web Interface - so is the Captive Portal just not finished yet? 
Are there any implementations available to get this working? 

I've also encountered a few small bugs that were fixed easily by myself, so I might open up a few pull requests later on. :-) 

Thanks for your work!

M1z23R, 2020-05-29:
I think that people missunderstand captive portal setting. It's just to enable the "pop-up interface" after you connect to the access point.

vuongfcc, 2020-06-06:
Maybe it will help you
https://www.hackster.io/244v234/esp8266-wifi-hacking-480f01

spacehuhn, 2020-06-06:
Captive portal is meant in a technical way here. It just means you will always be redirected to the web interface regardless of the requested address.
There are no extra files or specific user portal being served when the setting is enabled.

blackhacker3, 2020-07-17:
how to use captive portal feature  to redirect phishing site using esp8266

closed 2020-10-22 by stale[bot]

vuongfcc, 2020-11-03:
> cách sử dụng tính năng cổng bị khóa để chuyển hướng trang web lừa đảo bằng esp8266

https://www.hackster.io/234v244/esp8266-wifiphisher-4ed3d3

ahmedtowsikzap, 2022-02-12:
> how to use captive portal feature to redirect phishing site using esp8266

illegal stuff don't think about it! you will regret