SpacehuhnTech/esp8266_deauther #1237: Help with recreating deauth class

M1z23R, 2020-05-06:
Ok, so it's not actually an issue with the code or anything, I just thought it was better to ask here and/or at aircrack-ng git for help.

I am trying to create my own class that would do the following:
User inputs mac address of the AP with channel, and the class should create dauth packet for broadcast and send it. I am bumping my head here, so far I've learned from the code how deauth packet is made (updated) and that it's sent using sendPacket which is just a public bool in another class.
If someone could help me out create a function in this way I'd be very glad.
function (string ApBssid, int channe)
{
  uint8_t deauthPacket[26] = {
            /*  0 - 1  */ 0xC0, 0x00, 
            /*  2 - 3  */ 0x00, 0x00, 
            /*  4 - 9  */ 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
            /* 10 - 15 */ 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC,
            /* 16 - 21 */ 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC,
            /* 22 - 23 */ 0x00, 0x00,
            /* 24 - 25 */ 0x01, 0x00  };

sendPacket(deauthPacket);
}

I know it looks ugly but it's the best way i can explain it. Also, i've tried analyzing the code of aircrack-ng it's pretty much the same as here.

spacehuhn, 2020-05-06:
Maybe this part of the new code I am writing helps you a bit more: https://github.com/SpacehuhnTech/esp8266_deauther/blob/v3/esp8266_deauther/attack_deauth.h#L39

M1z23R, 2020-05-06:
Thank you so much for this, I'll try to work with it a little bit and see where it gets me.
Once again, thank you.

M1z23R, 2020-05-24:
As it turned out, I didn't really read entire tutorial on how to setup arduino ide for this.
I had newer version of esp8266 libraries and I didn't update the sdk using your files to allow raw packet sending. I've managed to figure everything out (at least the part of making my nodemcu deauth my own network - hard coded mac's). 
Thank you so much for your help, it took me a while but I am good to go now :)

closed 2020-05-24 by M1z23R