SpacehuhnTech/esp8266_deauther #1240: Can deauth_pkt be passed in as a serial port parameter？

shuimoshusheng, 2020-05-08:
Serial commands like attack -d deauth_pkt

spacehuhn, 2020-06-17:
Please elaborate on your question.
If you're talking about using the ESP8266 as a packet injector by sending the raw packets over serial, that's something on the todo for version 3.

shuimoshusheng, 2020-06-17:
> 请详细说明您的问题。
> 如果您正在谈论通过串行发送原始数据包来使用ESP8266作为数据包注入器，那么这就是版本3的任务。

thank you!Can the module receive the return packet of the injected data packet？like WiFi fuzzing test？

spacehuhn, 2020-06-17:
Sniffing is also on the todo for version 3 but the ESP8266 can't sniff the entire packet, at least the SDK doesn't allow you to. So only the first maybe 100 bytes of the packets are visible and could be analysed in wireshark.

closed 2020-09-22 by stale[bot]