SpacehuhnTech/esp8266_deauther #1244: Not accepting serial input from a Linux terminal

Ozymandias1700, 2020-05-16:
<details>
 I am running this firmware on a generic ESP-8266 ESP-01. Successfully flashed the firmware. When I connect to the chip over serial connection from a raspberry pi, all looks okay, I get what seems to be the normal startup info yet I cannot enter in any commands. I have tested this over ssh from two different machines and directly on the pi itself. 

The command I run is
`screen /dev/ttyAMA0 115200`
When I power up/power cycle the esp I get the following:
`Mounting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '9', hidden: false, captive-portal: true
STARTED! \o/
v2.1.0
# scan -t 5s
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
ID SSID                             Name             Ch RSSI Enc. Mac               Vendor   Selected
=====================================================================================================
 0 SKYI2YZA                                          11  -80 WPA2 a0:bd:cd:c6:0b:ea Bskyb            
=====================================================================================================
Stopped scan
Scan results saved in /scan.json
Starting Scan for stations (client devices) - 5s
Stopped Access Point
Scanning WiFi [60%]:   0 packets/s |  0 devices |  0 deauths
[===== Stations =====]
Station list is empty :(
Started AP
Stopped scan
Scan results saved in /scan.json
# sysinfo
[======== SYSTEM INFO ========]
RAM usage: 58880 bytes used [72%], 23040 bytes free [28%], 81920 bytes in total

Current WiFi channel: 9
AP MAC address: a6:cf:12:ef:de:96
Station MAC address: 00:0c:af:93:91:2e
SPIFFS: 3263 bytes used [1%], 230418 bytes free [98%], 233681 bytes in total
        block size 4096 bytes, page size 256 bytes
Files: 
  /names.json 2 bytes
  /ssids.json 318 bytes
  /autostart.txt 19 bytes
  /settings.json 503 bytes
  /scan.json 87 bytes
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '9', hidden: false, captive-portal: true
===============================
`

This all looks normal to me? Any reason it won't take any command input? If I try typing nothing shows up at all, all I can do is use normal screen commands using ctrl + a.

Ozymandias1700, 2020-05-17:
I don't know why, but it resolved itself.

closed 2020-05-17 by Ozymandias1700

spacehuhn, 2020-05-27:
For people finding this issue in the future - We're now working on a new serial terminal specifically for the deauther to make interacting over serial easier: https://github.com/SpacehuhnTech/Huhnitor