SpacehuhnTech/esp8266_deauther #1246: Deauth attack not working at all

anonbubble, 2020-05-20:
I have flashed the ESP successfully through COM1. I connected to that “pwned” network seamlessly with provided password. Now I visit the deauth.me and do all the scanning & selecting properly. I understand it only works with one 2.4GHZ network at a time with the 5GHZ one I disabled through my own router so it won’t interfere or switch. 

When I select my own network and go to the deauth attack, it says “1” target. Perfect. I press attack and nothing happens to my wifi. Meanwhile it’s showing 0/0 pkts to I hit reload, then it keeps showing me 20/20 most of the time. But everytime I switch pages to Settings or SSID and go back to the attack page, it always gives me a new packet number like 9/20, 12/20, 19/20 etc...

Long story short, It’s not doing anything to the wifi network. The beacon attack works perfectly and my cloned networks show up. I have also tried upping the packets to 25 then 255. Nothing. Also made the timeout 0, but it never attacked under 600 seconds anyway lol. I have searched the boards on here, YT and a good part of google trying to figure it out. Help please? This question isn’t formatted properly at all but I just made the account, hopefully this doesn’t ruin anybody’s year

<details>
  <summary> Compile log (...)</summary>

  

</details>
<br><br><br>

xcaiiox, 2020-06-09:
I also have this problem! I installed in five 8266 v3 model and the five are like this! Did you find any solution?

xcaiiox, 2020-06-09:
The model that was working, I reinstalled and stopped working! Now it's happening like the others.

xcaiiox, 2020-06-16:
It may not work anymore, I already bought more devices, tested it on different computers and routers. Nobody knows anything

spacehuhn, 2020-06-17:
Have you tried it out on other networks?

xcaiiox, 2020-07-16:
Any update? @anonbubble

closed 2020-10-21 by stale[bot]