SpacehuhnTech/esp8266_deauther #1247: REQUEST: (Better) Raspberry Pi support - Screens, Buttons, LEDs, etc.

Ozymandias1700, 2020-05-24:
### This is simply a suggestion!
The code works fantastically and I am so grateful to the dev for what they have done.

Put simply, the title sums this up. Currently, the firmware doesn't play too nicely with terminal emulators using a serial port (as another user raised in an issue). My current set up is:

- Pi zero
- ESP8266 (ESP01)
- (Planned) 0.96" LCD
- (Planned) Buttons for the Pi GPIO to control the deauther

Due to the current issue with this firmware and terminal emulators, I have written a custom script so I can send commands to the chip and view the output. It's messy and I'm sure someone else could do it better however I am but a noob. Adding gpio and custom screen support will not be easy for me however I will try my best, but if there was native linux/pi support I'm sure many people would be grateful.

I understand simply buying the NodeMCU or similar could bypass these problems but the Pi is a wonderful and versatile chip and if support could be added that would be fantastic.

Keep up the good work!

M1z23R, 2020-05-29:
Raspberry pi is much stronger device then nodemcu. It can run tools like aircrack-ng, mdk, etc.
But I think that this project is mainly a way to port tools like aircrack-ng onto a cheap, low powered device like nodemcu.

Ozymandias1700, 2020-05-30:
True, however I use the ESP as it means I can run these attacks etc while SSH'ing into the pi and not losing connection during monitor/injection as I can use the Pi's built in wifi in tandem with the ESP.

tobozo, 2020-05-31:
@Ozymandias1700 I've tested the [Pimoroni ESP IOT pHat](https://learn.pimoroni.com/tutorial/hal13/getting-started-with-iot-phat) on a Pi Zero and came up with two issues (specific to this build):

1) buttons aren't connected to the ESP but to the Raspberry Pi => requires extra logic
2) simultaneous use of both WiFi devices can trigger brownout problems on both sides => better use an external WiFi dongle, possibly with a 10+cm cable extension

[image]

M1z23R, 2020-05-31:
I've done a similar project with the pi using bluetooth for ssh, what capabilities of esp8266deauther would you like on your pi? Web interface, lcd/button support or something else?

Ozymandias1700, 2020-05-31:
Hey @tobozo , that's pretty impressive, nice work.
@M1z23R yeah pretty much what you mentioned, I know a lot of the things I would like are easily addressed by simply using the NodeMCU or other boards but however I have a few Pis and a few ESP-01s which as you know are the barest ESP you can get. That means to add screen and button support it will have to be all DIY as the ESP-01 has only the necessary pins, no extra GPIOs.
I think it would be useful to many to have more Pi support.

M1z23R, 2020-05-31:
Checkout this post for your PI.
You won't need bettercap, you can do the steps until you've done setting up bluetooth.
I might give it a go on porting this to RPI.

https://www.evilsocket.net/2018/07/28/Project-PITA-Writeup-build-a-mini-mass-deauther-using-bettercap-and-a-Raspberry-Pi-Zero-W/