SpacehuhnTech/esp8266_deauther #1249: Display doesn't work after compiling on Arduino IDE

TomEscolano, 2020-05-29:
Hi,

I just bought the DSTIKE Deauther watch on [Banggood](https://www.banggood.com/DSTIKE-OrangeBlack-Deauther-Wristband-Deauther-Watch-NodeMCU-ESP8266-Programmable-WiFi-Development-Board-p-1561694.htm).
I wanted to do a few tweaks to the source code and then upload it using the Arduino IDE.
I can successfully flash the watch using the .bin files,but even with proper configuration, the display doesn't boot up !
I modified the following values in my A_config.h file:
```
#define DEAUTHER_DISPLAY SH1106Spi display = SH1106Spi(5, 4, 15);
#define USE_DISPLAY true
```

I've read the Wiki a thousand times to find my mistake but I really struggle with it...

Thanks !

DangerousPony204, 2020-06-13:
i have the exact same problem with the same watch from the same place

DangerousPony204, 2020-06-13:
I found a solution for me, go to the web interface (192.168.4.1) when connected to pwned. then go under settings and check 'enable display'. It was turned off after compiling.

manishrd, 2020-07-23:
hi.. just received the watch. but after flashing through ESP 8266 flasher the RGB led is off. I want to go back to factory default setting in watch because. after fl;ashing successfully the LED turned off for ever and the navigation button is messed up. upon pressing it scrolls and on scroll it selects. please help how to get it to factory default

spacehuhn, 2020-07-23:
@manishrd try flashing the bin file, for example this one: https://github.com/SpacehuhnTech/esp8266_deauther/releases/download/2.5.0-rc1/esp8266_deauther_2.5.0_rc1_DSTIKE_DEAUTHER_WATCH.bin

satan666000, 2020-09-23:
In Arduino the same error when flashing and it does not turn on, so I recommend that you use your android phone and that is easier

closed 2020-12-29 by stale[bot]