SpacehuhnTech/esp8266_deauther #1251: Deauther OLED V6 - "resistor beside IPEX connector removing"

MrCook1es, 2020-06-03:
Good morning, I would like additional information about "resistor beside IPEX connector removing".
1) Was this resistance also present on the OLED V5 model?
2) The image below indicates the possibility of removing the resistance in order to obtain the "best effect" from the antenna. Why was the resistance included in the project? Is it because of the national regulations on the power Tx (mW)?
[image]

tobozo, 2020-06-03:
for a "best effect", nothing beats an amplifier

[image]

spacehuhn, 2020-06-17:
I think this is simply due to the use of the ESP-07 module. A common ESP8266 module that comes with the antenna connector but also a ceramic antenna. And if you remove the resistor, it will only be connected to IPX connector. Which means you **have** to have an external antenna connected (otherwise you can damange the device) but you also don't lose any energy to the less powerful ceramic antenna.
[image]

MrCook1es, 2020-06-17:
@spacehuhn thanks for the precise information.
I will be careful never to turn on my DSTIKE oled V5, without an external antenna connected.

there are different types of IPEX connectors, can you confirm that the one on your DSTIKE is an IPEX (MHF-4) type?
[image]

spacehuhn, 2020-06-17:
It's this one, don't ask me about the mess and confusion over the different connector standard and names
https://en.wikipedia.org/wiki/Hirose_U.FL

closed 2020-09-15 by MrCook1es