SpacehuhnTech/esp8266_deauther #1254: What exactly does the "Probe" attack do?

Iblis89, 2020-06-09:
I see that it has the same 60/60 SSID list indicator next to it that the Beacon attack uses, so I assume it has something to do with the saved SSID's but I can't find anything on the wiki offering an explanation of what precisely is occurring during a Probe attack. 

Thank you.

vuongfcc, 2020-06-13:
It attacks things like this [](https://www.manxtechgroup.com/people-counting-using-wifi-and-bluetooth/)

spacehuhn, 2020-06-17:
There is not info about it in the wiki, you're right that's something to work on.
But there is some info in the bottom of the attack page itself.

> Probe requests are sent by client devices to ask if a known network is nearby.
Use this attack to confuse WiFi trackers by asking for networks that you specified in the SSID list.
It's unlikely you will see any impact by this attack with your home network.

forseenink3238, 2020-06-17:
Hi, probe attacks I believe prevent WiFi handshakes

spacehuhn, 2020-06-17:
> Hi, probe attacks I believe prevent WiFi handshakes

Probe frames are used for an active network search but have nothing else to do with connecting to one. They do not prevent handshakes.

closed 2020-09-22 by stale[bot]