SpacehuhnTech/esp8266_deauther #1255: Deauth Attack not working :(

xcaiiox, 2020-06-10:
When I select my own network and go to the deauth attack, it says “1” target. Perfect. I press attack and nothing happens to my wifi. Meanwhile it’s showing 0/0 pkts to I hit reload, then it keeps showing me 20/20 most of the time. But everytime I switch pages to Settings or SSID and go back to the attack page, it always gives me a new packet number like 9/20, 12/20, 19/20 etc...

Long story short, It’s not doing anything to the wifi network. The beacon attack works perfectly and my cloned networks show up. I have also tried upping the packets to 25 then 255. Nothing. Also made the timeout 0, but it never attacked under 600 seconds anyway lol. I have searched the boards on here, YT and a good part of google trying to figure it out. Help please? This question isn’t formatted properly at all but I just made the account, hopefully this doesn’t ruin anybody’s year

jLynx, 2020-06-10:
Is your network 5g or 2.4g?

xcaiiox, 2020-06-10:
2.4

xcaiiox, 2020-06-10:
I have five units, the five are like this. The one that worked, I reinstalled to test if it was a problem in the other five and now it doesn't work either.

xcaiiox, 2020-06-10:
But I used the same files and the NODEMCU FLASHER

spacehuhn, 2020-06-17:
Have you tried it out on other networks?

xcaiiox, 2020-06-17:
Yes! I tested it on many networks and it didn't work on any :(

GrenderG, 2020-08-08:
Having the exact same issue, tried with different networks.

xcaiiox, 2020-08-09:
> Having the exact same issue, tried with different networks.

Unfortunately the device no longer works for me, I gave up a while ago.

Bommlon, 2020-08-16:
Try updating to the latest firmware. Maybe that helps.

xcaiiox, 2020-08-17:
> Try updating to the latest firmware. Maybe that helps.

Didn't work

Bommlon, 2020-08-31:
The deauther exploits a "loophole" in the way WiFi works. If you cant use the deauth attack on your network it means that your network is safe against it, which is at its core a good thing.

closed 2020-12-06 by stale[bot]

edisonbattery, 2020-12-09:
I just received mine today and Im having the exact same problem, so dont give us that "oh thats a good thing because your router blah blah blah"   I have one of your older ones and it works perfectly.   address the issue, sidestepping is lame.

RChadwick7, 2020-12-09:
Really? If you don't like it, ask for a refund.

edisonbattery, 2020-12-09:
I love your deauthers... but there is a problem with the dstike mini evo.
 I have an older version that works and the new one doesn't.

spacehuhn, 2021-01-14:
Made a few improvements the past days that should address this issue

Bakerboii, 2022-01-20:
I’m having the same issue. How do I fix it?

spacehuhn, 2022-02-06:
Hi, I know this issue is old and has been closed, but since it ranks well on Google, I would like to point people to a new blog post that discusses this topic in further detail:
https://blog.spacehuhn.com/deauth-attack-not-working/