SpacehuhnTech/esp8266_deauther #1257: No Deauth packets

ngtrung1992, 2020-06-14:
#https://github.com/SpacehuhnTech/esp8266_deauther/issues/1088
reopen my case
reproduce :
-I select one or multi APs
-On attack tab
-click start

( deauth method can not send any ptks but after symbol slash have number (Deauths: 0/40)
probes and beacons method still work )

[Pkt/s] All: 0 | Deauths: 0/40 | Beacons: 0/0 | Probes: 0/0
[Pkt/s] All: 0 | Deauths: 0/40 | Beacons: 0/0 | Probes: 0/0
[Pkt/s] All: 0 | Deauths: 0/40 | Beacons: 0/0 | Probes: 0/0
[Pkt/s] All: 0 | Deauths: 0/40 | Beacons: 0/0 | Probes: 0/0
[Pkt/s] All: 0 | Deauths: 0/40 | Beacons: 0/0 | Probes: 0/0

Issue when flash with my arduino 1.8.12 ( i just config settings.h file & installed board 2.1.0 ). no error show in when upload project to module.

Please reopen this case and tell me anything to solve.
p/s: No issue if auto flash with n2d2 tool

<br><br><br>

nGErWeigHtEMI, 2020-06-15:
it also can't send to detect stations

Scan results saved in /scan.json
Starting Scan for stations (client devices) - 15s
Stopped Access Point
Scanning WiFi [20%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [40%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [60%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [80%]:   0 packets/s |  0 devices |  0 deauths
[===== Stations =====]
Station list is empty :(

spacehuhn, 2020-06-17:
>  installed board 2.1.0

You need 2.0.0 or rather the Deauther SDK by Tobozo
https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

closed 2020-09-22 by stale[bot]